mocleiri/tensorflow-micropython-examples #27: Add Circuitpython compatibility

mocleiri, 2021-08-22:
We should be able to copy the code from ulab which also has circuitpython support.  It looks like you need to clone circuit python and then add a two links like this:

micropython-modules/microlite -> extmod/microlite
micropython-modules/audio_frontend -> extmod/audio_frontend

It would be similiar to this:
https://github.com/v923z/micropython-ulab/blob/1d9670096f093eae0a27437943dca088a27cc378/build-cp.sh#L41

We can try to use the version of ulab in circuit python but if that is a problem we can also make a symbolic link from:

micropython-ulab -> extmod/ulab

mocleiri, 2021-08-22:
We could implement a unix port build like ulab does but probably we would need to wait for #28 to be implemented before we could target an actual device as that will reduce the firmware size by allowing us to select less operators.

mocleiri, 2021-12-16:
The rp2 port works.  I should add esp32s2 support and then move forward with the circuitpython integration work.

uraich, 2025-09-29:
Yesterday I tried to recompile the esp32 port using the newest micropython version. I got a few compilation errors but adding MP_ERROR_TEXT in mp_raiseTypeError calls was all that was needed to build the system again. 
I did not try the camera and and the wake word examples yet but I managed to run the mnist hand-written digits example on the newly built system.