mocleiri/tensorflow-micropython-examples #41: i2c not working for esp32

AdoHaha, 2021-11-26:
Using the prebuilt firmware (https://github.com/mocleiri/tensorflow-micropython-examples/suites/4296460177/artifacts/112068306) 
the version Core Panic's for reading from I2C:

The simplest example to reproduce (assuming that there is the device on 0x53 addr, scl pin 18, sda pin 19):

```
import machine

from machine import ADC, Pin, I2C

a = i2c.readfrom_mem(0x53,50,6)
```

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008c868  PS      : 0x00060031  A0      : 0x8008c9fd  A1      : 0x3ffbe980  
A2      : 0x3ffb6fd8  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000800  
A6      : 0x3ffbf008  A7      : 0x00000000  A8      : 0x3ffb71e4  A9      : 0x00000901  
A10     : 0x000000a5  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffb71e8  
A14     : 0x00000000  A15     : 0x3ff53000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4008c865:0x3ffbe9800x4008c9fa:0x3ffbe9c0 0x40082d51:0x3ffbe9f0 0x40241a9f:0x3ffbcfe0 0x40122d57:0x3ffbd000 0x40096344:0x3ffbd020 

ELF file SHA256: 84e9af058336fb6e

AdoHaha, 2021-11-27:
I also verified this with second hardware peripheral (i.e. I2C(1 ... )) as well as different pins

The initialization works, as well as scanning (i2c.scan()) and sending to the i2c device. 
 It crashes on reading from the device (readfrom_mem, readfom_mem_into ...)

AdoHaha, 2021-11-27:
I use the normal (non spiram ) version of the firmware as my boards do not have external RAM. I found this error: https://github.com/espressif/esp-idf/issues/6336 which seems to the situation when SPI ram is connected and for particular pins, in my case, the error occurs on various pin combinations. I switched to soft I2C but I think there is the possibility of some larger bug connected to address space

uraich, 2021-12-02:
I had the same problem when using a very new version of esp-idf (v4.4 or newer). It is a bug in esp-idf. If you build micopython with esp-idf-v4.3, the currently latest stable version, the problem does not arise. 
Here is the modification I had to make in components/driver/i2c.c at line 1343:
            if (i2c_cmd_is_single_byte(cmd) || cmd->total_bytes == cmd->bytes_used) {
                p_i2c->cmd_link.head = p_i2c->cmd_link.head->next;
		//                p_i2c->cmd_link.head->cmd.bytes_used = 0;
		if(p_i2c->cmd_link.head!=NULL)
		  p_i2c->cmd_link.head->cmd.bytes_used = 0;
            }
The original line is the one commented out.

mocleiri, 2021-12-05:
Thanks.  

@AdoHaha the code on the main branch is already building with esp-idf v4.3 but I changed the code on more in development builds to use esp-idf v4.3.1 instead.  

Can you try the updated firmware here: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1540331954

AdoHaha, 2021-12-05:
Hi, thanks for the update. The core panic no longer appears and hardware I2C works correctly.

closed 2021-12-06 by AdoHaha