mocleiri/tensorflow-micropython-examples #42: Problem compiling tensorflow library

uraich, 2021-12-01:
tensorflow/lite/micro/kernels/l2_pool_2d.cc:128:1: error: insn does not satisfy its constraints:
which xtensa compiler version do you use?

TGiles1998, 2021-12-03:
tensorflow/lite/micro/kernels/l2_pool_2d.cc:128:1: error: insn does not satisfy its constraints:

I have experienced the same error, will try a different xtensa compiler version

uraich, 2021-12-04:
I investigated the problem a little further, and I found that another compiler and/or esp-idf version does not help. However, there are build scripts in 'ci'. I simply had to modify the BASE_DIR to point to my installation directory in build_tensorflow_esp32.sh. Then the tensorflow library compiles ok. 
I had to run refresh-tensorflow-microfrontend.sh in micropython-modules/audio_frontend to create fft.cpp, which was otherwise missing.
Then I replaced micropython with the latest version, which I cloned from its git repository, and finally I finally managed to build with build_micropython_esp32_microlite.sh.
You will find the build directory in boards/esp32/MICROLITE/build

TGiles1998, 2021-12-04:
Thank you will give this a go!

mocleiri, 2021-12-05:
From the comments @uraich left I think you are building from the main branch where there are two builds:
1. for tensorflow to a static library
2. for micropython with the microlite and audio_frontend modules.

I haven't build from main in a while which is probably the cause here. 

Micropython added some optimizations which caused the build to run at the -O2 optimization level.  That is what triggers the isn error.  I filed an issue upstream in https://github.com/tensorflow/tflite-micro/issues/634 to see if they can change the reference implementation to avoid the error.

But I think a better way to fix is to either change the esp32 build so that it will use -O3 only for that file or to add in some esp32 specific kernels using the esp-dsp library into upstream tensorflow micro, see: #9.

The current experimental work changes the entire build to -O3 to get the build to work.

I'll see if I can find an example of how to change a cmake build to use different optimization options for a single file.

uraich, 2021-12-07:
Hi Michael,
First of all, l a big "THANK YOU" ! Your work is extremely helpful for me to learn something about TinyML. 
As I said before, I managed to get everything compiled, and I'm about trying the examples. For the moment, I only had a look at helloworld using the tflite model you provide. There, everything works fine, which means that MicroPython sees the tflite library and the Python interface to it. I will now try to use the output of the model to control a LED, as shown in some YouTube videos.
After that, I will attack the microphone example.

closed 2021-12-08 by uraich