mocleiri/tensorflow-micropython-examples #48: Custom ESP32-S3 External RAM Configuration

TGiles1998, 2021-12-18:
Following on from the issue: https://github.com/mocleiri/tensorflow-micropython-examples/issues/40

Error produced when trying to run the Micro-Speech example on the ESP32-S3 was:

Connecting to /dev/tty.SLAB_USBtoUART...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf60
load:0x403b6000,len:0x978
load:0x403ba000,len:0x2c80
entry 0x403b616c
W (25) bootloader_random: RNG for ESP32-S3 not currently supported
W (313) bootloader_random: RNG for ESP32-S3 not currently supported
Traceback (most recent call last):
  File "main.py", line 14, in <module>
ImportError: no module named 'audio_frontend'
MicroPython 59e6194 on 2021-12-18; ESP32S3 module (microlite-spiram) with ESP32S3
Type "help()" for more information.

I'll also look into the RNG errors produced as well.

mocleiri, 2021-12-19:
@TGiles1998 

That error is most likely due to my moving audio_frontend from a separate module into another type in microlite.  

Try 
''''
af = microlite.audio_frontend()

af.configure()

af.execute(...)
'''

I updated the main.py and micro_speech.py here to use the new location:
https://github.com/mocleiri/tensorflow-micropython-examples/tree/main/examples/micro-speech/esp32

The model.tflite comes from here:
https://github.com/mocleiri/tensorflow-micropython-examples/tree/main/examples/micro-speech/lib

This is a short video of the example running for me:
https://youtu.be/FBUwIvzNVd4

That video was the double check that micro-speech still ran after the audio_frontend was integrated.

You need to change the I2S pins in main.py to match how you connected the microphone.

There is definitely room for improvement in the examples so please continue to report any problems you find with them so I can improve things.

TGiles1998, 2021-12-20:
That patch seems to work thanks! I will continue to test and expand on these examples.

Would it be possible to further customize the GPIO 20 CS build to allow for the full 8MB of  SPI-RAM to be used? I believe it just requires the ESP32S3 specific config options in the "idf.py menuconfig" to be selected with the CONFIG_SPIRAM_TYPE_ESPPSRAM64.

I think I put the config options here: https://github.com/mocleiri/tensorflow-micropython-examples/issues/40#issuecomment-989323462

Thanks so much!

mocleiri, 2021-12-20:
Can you report what micropython.mem_info() shows ?

Unexpected Maker flashed essentially the same config but without the spiram size given and it found all 8MB automatically.  Start at the 15 minute mark: https://youtu.be/rcTrYBtaAMc

mocleiri, 2021-12-23:
@TGiles1998 I went back and applied all the options you reported originally on #40.

This built locally for me so I think it will via the github actions aswell.

https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1614197339

I left the CPU frequency at 240 MHz but all of the options you had should be applied now.

TGiles1998, 2021-12-23:
@mocleiri 

Sorry for the late reply! Thank you that's fantastic! Will start testing this!

mocleiri, 2022-01-19:
@TGiles1998 I will close this issue.  Please open others if you want further adjustments made to the configs or anything else.

closed 2022-01-19 by mocleiri