mocleiri/tensorflow-micropython-examples #53: Build for STM32 F446RE Board

pythonbobo, 2022-01-03:
Hi,
I am very interested in your project and just want to answer to your README.
Is it possible to add an build for the nucleo F446RE Board?

Greetings

mocleiri, 2022-01-04:
I can base the microlite build on the upstream NUCLEO_F446RE board:
https://github.com/micropython/micropython/tree/master/ports/stm32/boards/NUCLEO_F446RE

mocleiri, 2022-01-04:
I have a partial build working but there are some problems.

Are these the specs for your board?

180 MHz max CPU frequency
512 KB Flash
128 KB SRAM System

source: https://os.mbed.com/platforms/ST-Nucleo-F446RE/

How much heap do you see when running stock micropython?  It may be enough if you have around 50 kb.  

The main issue is firmware size:
```
arm-none-eabi-ld: /home/mike/git/tensorflow-micropython-examples/boards/stm32/NUCLEO_F446RE_MICROLITE/build/firmware.elf section `.isr_vector' will not fit in region `FLASH_ISR'
arm-none-eabi-ld: /home/mike/git/tensorflow-micropython-examples/boards/stm32/NUCLEO_F446RE_MICROLITE/build/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
arm-none-eabi-ld: region `FLASH_ISR' overflowed by 31532 bytes
arm-none-eabi-ld: region `FLASH_TEXT' overflowed by 5658932 bytes
```

The firmware fits in my 2 MB flash partition on my H743ZI2 but it looks like 512kb is not enough.

The main solution is to come back to this board after #28 is merged which allows for compile time selection of which ops are included in the firmware.  The core tensorflow runtime is supposed to be on the order of 20 kb so excluding most of them should work to fit the firmware to your board.

mocleiri, 2022-01-07:
I have  #28  working on main but even when building without the math or c++ library I still get over by 5.5 MB.
```
build/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
arm-none-eabi-ld: region `FLASH_ISR' overflowed by 31532 bytes
arm-none-eabi-ld: region `FLASH_TEXT' overflowed by 5507560 bytes
```

I need to compare the sizes that are reported on the standard micropython build to see how much I need to trim and what is contributing the most to the firmware  size.

pythonbobo, 2022-01-07:
Sry for my late answer:
`How much heap do you see when running stock micropython?`
Normally I do not really consider about this. For ordinary STM Projects I use the default values with min. heap size:0x200 and min. stack size:0x400

I hope this can help you.

Best Regards

mocleiri, 2022-01-11:
I've been doing some testing.  I think the H743ZI2 build is about 650 KB in size.  I'm trying to lower it down.

I've been playing with #28 applied to this problem but link time optimization is clearly dropping functions but its not dramatically lowering down the firmware size.

I will keep this open but I suspect at the moment you need to use an stm32 board with at least 1MB of flash.

mocleiri, 2022-01-11:
I disabled all of ulab and all of tensorflow lite micro and got a firmware size of 360k.  I need to take a closer look at the ops needed for the examples to see if I can get a firmware that will have enough ops within it to do anything.
``` 
text    data     bss     dec     hex filename
 360944      88   37820  398852   61604 /home/mike/git/tensorflow-micropython-examples/boards/stm32/NUCLEO_F446RE_MICROLITE/build/firmware.elf
```

mocleiri, 2022-01-17:
@pythonbobo can you download the firmware for the F446RE from here: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1699763892

I don't think this version is going to work but I wonder if you can report what you see and what fails when you load it.  If it does get to the repl port please copy the [hello-world example model and python code](https://github.com/mocleiri/tensorflow-micropython-examples/tree/main/examples/hello-world) over and see if it will run.

I merged code for #28 that allows for selecting which tflm op is being used but its not working quite like I think it should work.

You can't see it in the build output but there are linking errors that I'm ignoring to get a firmware file generated.

But the only place calling the missing functions are from code blocks that are being selectively included from a custom op resolver so its possible that it will allow you to run the firmware.

mocleiri, 2022-01-19:
@pythonbobo #28 isn't working right.  I had to revert it and will revisit in #63 I don't think the image I linked to is going to work for you.