mocleiri/tensorflow-micropython-examples #55: Playing on micro-speech

uraich, 2022-01-04:
I saw that the new unix build has been merged into main. I tried it and was successful.
Now I am trying to understand the micro-speech example. Since you have integrated the audio front end into the microlite module the example code does not work any more.
I added:
audio_frontend = microlite.audio_frontend()  # before 
audio_frontend.configure()
in main.py and micro_speech.py to get that part to work.
However I still get an error when trying to create the interpreter:
[image]

mocleiri, 2022-01-04:
Thanks for this report.  The unix port was the first one that ran micro_speech but is now very out of date.

I plan on consolidating the different port scripts into a single script that will work for the different ports. See #50.

Can you try the scripts from the esp32 or rp2 directories stripping out the I2S part?

I'm going to fix this one by using the unix port to run the examples as part of the CI build process.

uraich, 2022-01-04:
On the ESP32 version I actually get the same problem:
[image]

mocleiri, 2022-01-05:
@uraich the model is wrong.  Note the UINT8 part.  This is an old model that I had from the beginning but it doesn't work.  In fact tflm has removed support for UINT8 altogether which is why you get those strange errors.

I filed an upstream documentation fix about this since the model file itself was distributed in a zip file which needed to be updated:
https://github.com/tensorflow/tensorflow/issues/48752

I have this working locally and will commit soon.

mocleiri, 2022-01-05:
I've committed the changes (https://github.com/mocleiri/tensorflow-micropython-examples/pull/58). Can you see if they solve the problem for you?

For ESP32 double check that you are using the model.tflite from the examples/micro-speech/esp32 directory.

uraich, 2022-01-05:
I pulled the new repo and tried the unix version, which now works fine for me (same output as shown above)
You say I should use model.tflite from the examples/micro-speech/esp32 directory but there is no model file. I guess that model.tflite from examples/micro-speech/unix should be the version I will have to upload to the ESP32.
I don't have the microphone going yet, but the error when creating the interpreter has gone.
I will now play on the microphone and as soon as I have that working I will try again.

mocleiri, 2022-01-05:
The model file to use is currently here in examples/micro-speech/lib: https://github.com/mocleiri/tensorflow-micropython-examples/blob/main/examples/micro-speech/lib/model.tflite

I will fix this mess in #50 where there will be just one model file and one set of files.