mocleiri/tensorflow-micropython-examples #60: Building ESP32S3 Firmware VS Code Mac Terminal

TGiles1998, 2022-01-13:
The following commands were run on macOS Monterey - version 12.1 Terminal on VS Code

git clone https://github.com/mocleiri/tensorflow-micropython-examples.git

# Then Following the build_esp32s3.yml in the .github/workflows folder.

cd /tensorflow-micropython-examples

git submodule init
git submodule update --recursive
cd micropython
git submodule update --init lib/axtls
git submodule update --init lib/berkeley-db-1.xx
cd ..

IDF_COMMIT=142bb32c50fa9875b8b69fa539a2d59559460d72
echo "esp-idf-commit=$IDF_COMMIT" >> $GITHUB_ENV
TFLM_COMMIT=$(git submodule status tensorflow | awk '{print ($1)}')
echo "tflm-commit=$TFLM_COMMIT" >> $GITHUB_ENV

source ./micropython/tools/ci.sh && ci_esp32_idf44_setup

source ./esp-idf/export.sh

pip3 install Pillow
pip3 install Wave

echo "Regenerating microlite/tfm directory"
rm -rf ./micropython-modules/microlite/tflm

cd ./tensorflow

../micropython-modules/microlite/prepare-tflm-esp.sh

cd ..

source ./esp-idf/export.sh

cd ./micropython

echo "make -C mpy-cross V=1 clean all"

make -C mpy-cross V=1 clean all

cd ..

echo "cd ./boards/esp32/MICROLITE_S3"
cd ./boards/esp32/MICROLITE_S3

echo "Building MICROLITE_S3"
rm -rf build
idf.py clean build

FAILED: 

/tensorflow-micropython-examples/micropython/ports/esp32/main.c:38:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.
/tensorflow-micropython-examples/micropython/ports/esp32/gccollect.c:37:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"
          ^~~~~~~~~~~
compilation terminated.
/tensorflow-micropython-examples/micropython/ports/esp32/modmachine.c:46:10: fatal error: esp32s3/clk.h: No such file or directory
 #include "esp32s3/clk.h"
          ^~~~~~~~~~~~~~~
compilation terminated.

returned non-zero exit status 1.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

I can find all of these files by hand in the downloaded ESP-IDF in the repository but the compiler just can't seem to pick them up. any suggestions would be fantastic thank you!

TGiles1998, 2022-01-14:
Copying and pasting the missing individual folder and files into the board root folder does seem to partially solve it but then produces further errors:

[1116/1606] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/...ensorflow-micropython-examples/micropython/extmod/moduwebsocket.c.ob
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
[1118/1606] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/.../tensorflow-micropython-examples/micropython/extmod/uos_dupterm.c.ob
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
[1120/1606] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/...1/tensorflow-micropython-examples/micropython/extmod/modwebrepl.c.ob
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
[1121/1606] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/.../tensorflow-micropython-examples/micropython/extmod/utime_mphal.c.ob
cc1: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
...

etc for all of the remaining build lines

mocleiri, 2022-01-14:
My development environment is windows but I'm doing most work in visual studio code running in windows subsystem for linux so I'm unable to help directly.

However if you are able to figure out the steps to build on MacOS please feel free to contribute the steps and we can improve the manual build instructions to include for mac os.

The -fno-rtti error is not an error per-se.  I need to improve the microlite cmake file to seperate the C++ build from the C build.
CMake is passing that option to gcc which is complaining that its not a C option.  But I added for the C++ code where it is needed.

That line is emitted right now for all esp32 builds in linux.

If you get a firmware generated at the end of the build then I think you have a working build.

TGiles1998, 2022-01-15:
Thank you for your reply! I will focus on the mac build and reply back to the thread so it can be added. Thanks again!

TGiles1998, 2022-01-28:
Solved: Ive solved it using latest IDF-4.4 and making sure all of the toolchain had been properly added to the bash Profile. Ive added a pull request with a .md document detailing the process.

The Process is as pasted below:

# Build Tensorflow Micropython on MAC OS using Bash Terminal

Tested on: macOS Monterey Version 12.1

## Prerequisites 

# Properly Installing ESP-IDF (Taken from: https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/macos-setup.html)

sudo easy_install pip

pip install --user pyserial

brew install cmake ninja

brew install ccache

Download: https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

mkdir -p ~/esp
cd ~/esp
tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

# Make Toolchain Accessible from Bash Terminal 

Open your /bash.profile file by navigating to usr directory and pressing down the keys "Command, Shift, FullStop" it shoudl appear.

Add to your profile:

export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
alias get_esp32="export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH"

# Git Clone ESP-IDF Version 4.4

cd ~/esp
git clone -b v4.4 --recursive https://github.com/espressif/esp-idf.git esp-idf

# Update the Submodules

cd esp-idf
git submodule update --init

# Reopen your /bash.profile and add the following:

export IDF_PATH=~/esp/esp-idf
export PATH="$IDF_PATH/tools:$PATH"

# Install the required python packages 

python3 -m pip install --user -r $IDF_PATH/requirements.txt

## Download the Tensorflow Micropython Examples Repo

git clone https://github.com/mocleiri/tensorflow-micropython-examples.git

## Building ESP32S3 

# Building the TensorFlow Firmware

cd /tensorflow-micropython-examples

git submodule init
git submodule update --recursive
cd micropython
git submodule update --init lib/axtls
git submodule update --init lib/berkeley-db-1.xx
cd ..

# Get Cache Keys

IDF_COMMIT=142bb32c50fa9875b8b69fa539a2d59559460d72
TFLM_COMMIT=$(git submodule status tensorflow | awk '{print ($1)}')

# Setup IDF within Repo (Another Option)

source ./micropython/tools/ci.sh && ci_esp32_idf44_setup

source ./esp-idf/export.sh

pip3 install Pillow
pip3 install Wave

rm -rf ./micropython-modules/microlite/tflm

cd ./tensorflow

../micropython-modules/microlite/prepare-tflm-esp.sh

cd ..

# Build Micropython Cross-Compiler 

source ./esp-idf/export.sh
cd ./micropython
make -C mpy-cross V=1 clean all

cd ..

# Build Generic ESP32S3 8MB FLASH

source ./esp-idf/export.sh 

(OR: get_esp32
get_idf)

cd ./boards/esp32/MICROLITE_S3

rm -rf build
idf.py clean build

# To build other boards e.g. GENERIC_SPIRAM replace with Board definition

cd ./boards/esp32/MICROLITE_S3_SPIRAM

If succesful you should see a reult similar to this:

/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x000000    18640  (   14128 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1191120  (  840496 remaining)
total                  1256656

mocleiri, 2022-01-30:
The corresponding pull request for this was #66 it was merged applied in d083f23ad32a16bf40b7383a69bf98ab1c6c9281.

closed 2022-01-30 by mocleiri