mocleiri/tensorflow-micropython-examples #61: esp32-s3 images don't work

DaniDD, 2022-01-18:
Hi,
Im testing any from the last prebuild images for ESP32-S3, but no one work. E.g.: this: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1699763888
I have a reboot loop after 30sec. In an post I have read I must disable USB in mpconfigboard.cmake
After I remove the line with USB, my build image does work.

mocleiri, 2022-01-19:
I made the change to both the MICROLITE_S3 and MICROLITE_S3_SPIRAM boards.  Can you download from here and confirm they work?
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1716379308

DaniDD, 2022-01-19:
I try this build for S3:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1716379305
It works now. Thank you, great job!

mocleiri, 2022-01-25:
Closing as the changes have fixed the problem.

closed 2022-01-25 by mocleiri