mocleiri/tensorflow-micropython-examples #62: esp32-s3 hello-world does not work

DaniDD, 2022-01-18:
Hello,

I try the hello-world example on ESP32-S3. If I start the example, I have this error, then Guru Mediatation and reboot. My output:
`
interpreter_make_new: model size = 2488, tensor area = 2064
12:31:04.012 -> Didn't find op for builtin opcode 'FULLY_CONNECTED' version '4'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?
12:31:04.012 -> 
12:31:04.012 -> Failed to get registration from op code FULLY_CONNECTED
12:31:04.012 ->  

12:31:04.012 -> AllocateTensors() failed!
12:31:04.012 -> time step,y
12:31:04.012 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
12:31:04.012 -> 
12:31:04.012 -> Core  0 register dump:
12:31:04.012 -> PC      : 0x4200c6db  PS      : 0x00060730  A0      : 0x82001b90  A1      : 0x3fce4390  
12:31:04.059 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fca3bf8  A5      : 0x00000314  
12:31:04.059 -> A6      : 0x00000090  A7      : 0x3fca5ba0  A8      : 0x8209cfb4  A9      : 0x3fce4380  
12:31:04.059 -> A10     : 0x3fca64a0  A11     : 0x00000000  A12     : 0x00000010  A13     : 0x3fca64b0  
12:31:04.059 -> A14     : 0x00000030  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
12:31:04.059 -> EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
12:31:04.059 -> 
12:31:04.059 -> 
12:31:04.059 -> Backtrace:0x4200c6d8:0x3fce43900x42001b8d:0x3fce43b0 0x4200183d:0x3fce43d0 0x4205f096:0x3fce43f0 0x420672dd:0x3fce4410 0x40378709:0x3fce4430 0x4205f190:0x3fce44d0 0x4206726d:0x3fce4540 0x42001c11:0x3fce4570 0x42001a42:0x3fce4590 0x4205f079:0x3fce45b0 0x420672dd:0x3fce45d0 0x40378709:0x3fce45f0 0x4205f190:0x3fce4690 0x420673a9:0x3fce46f0 0x420ab2ed:0x3fce4780 0x4205f1ff:0x3fce47b0 0x4206721d:0x3fce47e0 0x4037867d:0x3fce4800 0x4205f190:0x3fce48a0 0x42067245:0x3fce48f0 0x420a84d7:0x3fce4910 0x420a88ed:0x3fce49a0 0x42082ee3:0x3fce49d0 
12:31:04.107 -> 
12:31:04.107 -> 
12:31:04.107 -> 
12:31:04.107 -> 
12:31:04.107 -> ELF file SHA256: 1ecabd06b77e03b1
12:31:04.154 -> 
12:31:04.154 -> Rebooting...
12:31:04.154 -> 
...
`

mocleiri, 2022-01-19:
Thanks for this report.  I'm seeing the same thing for regular esp32 builds.  

I merged #28 as it seemed to be working but it looks like it was just compiling but not actually doing what it is supposed to which is to allow controlling which tflm ops are included in the firmware.

I'm tracking the fix for this on #63 for now I think you should try using this version which is before I merged #28:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1656246622 (at commit: c98c091ce686bf14af9f505ce3ff75ed4cc81a9b)

mocleiri, 2022-01-19:
I reverted the merge of #28 and you should be able to download the updated board image form here:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1716379308

DaniDD, 2022-01-19:
Thank you mocleiri für your fixes. The firmware for ESP32_S3 booting now and I can run hello_world.
I have test ist with this build: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1716379305

mocleiri, 2022-01-25:
Closing as the changes have fixed the problem.

closed 2022-01-25 by mocleiri