mocleiri/tensorflow-micropython-examples #69: Using without callbacks for input and output?

cgreening, 2022-01-30:
I'm fairly new to Micropython, so this might not be the way things work.

Is it possible to use the interpreter without input and output callbacks?

e.g.

```
model = bytearray(59760)
model_file = io.open('test.tflite', 'rb')
model_file.readinto(model)
model_file.close()

interp = microlite.interpreter(model, 4096, None, None)
inputTensor = microlite_interpreter.getInputTensor(0)
# fill the input tensor with data
outputTensor = microlite_interpreter.getOutputTensor(0)
interp.invoke()
result = outputTensor.getValue(0)
```

mocleiri, 2022-01-30:
The callback approach is not the best and dates from the very beginning (January 2021) when I didn't know how anything worked.

I plan to change how the interpreter works in #10 to try and line up as a subset of the tensorflow lite python  Interpreter API.

I just took a look at your question on my m5 camera board and its possible to not use the callbacks but at the moment you have to stub a specific kind of no-op function.

```
def noop(interp):
    pass

interp = microlite.interpreter(model, 4096, noop, noop)
inputTensor = microlite_interpreter.getInputTensor(0)
# fill the input tensor with data
outputTensor = microlite_interpreter.getOutputTensor(0)
interp.invoke()
result = outputTensor.getValue(0)
```

I'll make an API change on this issue so that if you don't set callbacks don't use them.  

i.e. 

```
interp = microlite.interpreter(model, 4096)
inputTensor = microlite_interpreter.getInputTensor(0)
# fill the input tensor with data
outputTensor = microlite_interpreter.getOutputTensor(0)
interp.invoke()
result = outputTensor.getValue(0)
```

It seemed like it was a little bit faster without the function calls.

cgreening, 2022-01-30:
That works nicely!

closed 2022-01-30 by cgreening