mocleiri/tensorflow-micropython-examples #70: Build for NUCLEO_F746ZG - FATAL ERROR MemManage when running hello_world

jpliew, 2022-01-31:
Hi all,

I have created a build based on the STM32 samples from this repo for [NUCLEO_F746ZG with STM32F746ZG](https://www.st.com/en/evaluation-tools/nucleo-f746zg.html). Using a good [commit 49521](https://github.com/mocleiri/tensorflow-micropython-examples/tree/feature/63-fix-configurable-op-resolver) the compilation and generation of the firmware was successful but when executing the `hello_world.py`, there was FATAL ERROR MemManage and no more response.

## Firmware.elf
```
 arm-none-eabi-size build/firmware.elf
   text    data     bss     dec     hex filename
 715072     284   83924  799280   c3230 build/firmware.elf
```

## Terminal Output
```
>>> MicroPython v1.17-333-gcf258c898-dirty on 2022-01-31; NUCLEO_F746ZG MICROLITE with STM32F746
Type "help()" for more information.
>>> import hello_world.py
interpreter_make_new: model size = 2488, tensor area = 2064

FATAL ERROR:
            MemManage
```

## HEAP from Current Micropyhton release
```
>>> MicroPython v1.18 on 2022-01-17; NUCLEO-F746ZG with STM32F746
Type "help()" for more information.
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
163920
```

## HEAP from this build
```
MicroPython v1.17-333-gcf258c898-dirty on 2022-01-31; NUCLEO_F746ZG MICROLITE with STM32F746
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
158576
```

Is there anything else that I can provide for someone to check and provide suggestions on how to debug this ?

Thanks

mocleiri, 2022-02-01:
@jpliew thanks for this report.  

I did have a working build for STM32H743ZI2 which worked for hello-world but not micro-speech because there were HAL differences.

I can tell you that the build you are using is definitely not going to work because its from a point in time where the build worked but the program didn't work. 

I'm going to spend some time to get the STM32 build working again and/or identify the historical build that was working and you can setup your board at that commit.

the current problem is that the image size is too big.  I have a STM32F405 with 16 MB flash that I plan to use to validate if the code is working and then from there I will find out where the extra size is coming from and remove it.

jpliew, 2022-02-01:
Thanks @mocleiri , if you need me to test anything, please let me know.

mocleiri, 2022-02-25:
@jpliew sorry for the delay in fixing this.  My first pass I tried to get the builds working and optional ops working but while I got everything compiling it didń´t work when flashed to the device.  

I ended up just fixing the build issues for stm32 in #73.  

If you rebase your board configs to the latest commit on the main branch you should be able to compile working firmware for the F746ZG board.

The key ingredient to a working build is to  add this option to the mpconfigboard.mk file:
```
MICROPY_ROM_TEXT_COMPRESSION = 0
```

I was able to flash the h743zi2 firmware and run hello_world successfully: 
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1896970048

jpliew, 2022-02-25:
Thanks @mocleiri , I will try it and let you know the outcome.

Cheers