mocleiri/tensorflow-micropython-examples #75: Using TensorFlow with MicroPython on the Pico???

Raunak-Singh-Inventor, 2022-03-08:
Does this repo currently support using TensorFlow with MicroPython on the Raspberry Pi Pico? If not, is support going to be added in the near future?

mocleiri, 2022-03-13:
@Raunak-Singh-Inventor yes there are rp2 build's downloadable from here:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1904374806

Carn you check if your particular pico board is being auto built?

I'd be happy to add your board of it's not already being built.

I'm building using the cmsis accelerated kernels and have run hello world and micro speech on the pico.

Raunak-Singh-Inventor, 2022-03-14:
@mocleiri I am using the ArduCam Pico4ML board: https://www.arducam.com/pico4ml-an-rp2040-based-platform-for-tiny-machine-learning. Would it be possible to add builds for this pico board?

mocleiri, 2022-03-15:
@Raunak-Singh-Inventor I can't find any board specific pins or setup either at that link or within the pick-sdk for your board.  This makes me think that the standard pico build will work.

Can you try and flash [microlite-rp2-pico-firmware](https://github.com/mocleiri/tensorflow-micropython-examples/suites/5441404070/artifacts/172700732).

Hello world and micro_speech should work.  

At the moment the person detection has only been written for esp32 but it looks like arducam have a micropython driver for the camera: 
https://github.com/ArduCAM/PICO_SPI_CAM/tree/master/Python

That could be used to replace the camera module in the esp32 example: https://github.com/mocleiri/tensorflow-micropython-examples/blob/main/examples/person_detection/esp32-cam/person_detection_cam.py

Raunak-Singh-Inventor, 2022-03-17:
@mocleiri I uploaded the `firmware.uf2` to the pico. Now, how do I run the hello_world example?

Raunak-Singh-Inventor, 2022-03-17:
Never mind, will check out these instructions https://github.com/mocleiri/tensorflow-micropython-examples/tree/main/examples/hello-world#rp2-port

closed 2022-03-17 by Raunak-Singh-Inventor