mocleiri/tensorflow-micropython-examples #77: Error in import microlite

LAMSTREAM, 2022-03-15:
Hi,mocleiri!
I have learn about arduino and try to play micropython recently
I am using the rasdpberrry pico and upload the firmware successfully
But when I import hello_world.py,it doesn't work
cross some efforts,I find that I can import the ulab but microlite can not
can you help me to solve this problem?
THANKS!

mocleiri, 2022-03-15:
@LAMSTREAM can you tell me which firmware you flashed?  

The latest from the main branch works but there are other builds on other branches that don't work right.

These are the latest pico main branch builds: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1896970046

LAMSTREAM, 2022-03-15:
> @LAMSTREAM can you tell me which firmware you flashed?
> 
> The latest from the main branch works but there are other builds on other branches that don't work right.
> 
> These are the latest pico main branch builds: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1896970046

I have try this firmware and also #79 firmware,they both didn't work :(
[https://imgur.com/a/7w8dbrT](url)

mocleiri, 2022-03-15:
I have an [adafruit feather rp2040](https://learn.adafruit.com/adafruit-feather-rp2040-pico)  I'll run the latest and see what happens for me.  I've never had any issues with the firmware its always just worked.  I have seen lots of crashes in the stm32 build when invoking the interpreter but never I don't think in just loading the module.

What specific board do you have?  I think there should be a way to turn on verbose error messages to try to see/understand what the error is that is causing the board to hang.

LAMSTREAM, 2022-03-19:
Really thank you my mocleiri！
I solve this problem
My board is raspberry official board,and the firmware for it are really doesn't work
But I try to upload the Adaruit feather-rp2040's firmware for it this night.
and it truly work now!
Thanks you work for this!

mocleiri, 2022-03-21:
I can see that my board is currently built first I'll reorder and see if I can reproduce the issue. Thanks for reporting and also how you fixed it