mocleiri/tensorflow-micropython-examples #78: ESP official ESP32-S3-EYE v2.2 development board flash firmware

pyharry, 2022-03-15:
Hello! Please tell me about the situation of ESP official ESP32-S3-EYE v2.2 development board flashing firmware (development board link: https://github.com/espressif/esp-who/blob/master/docs/en/get- started/ESP32-S3-EYE_Getting_Started_Guide.md), I tried to flash microlite-esp32-s3-firmware; microlite-esp32-s3-spiram-cs-gpio-20-firmware; microlite-esp32-s3-spiram-firmware; microlite -esp32-s3-spiram-octal-firmware, only one line is returned in PuTTY, there is no Micropython prompt to return, it seems to be stuck there, I don't know where is the problem? Thanks! As shown below
[image]
[image]
[image]
[image]
[image]
[image]

pyharry, 2022-03-16:
Hello, I tried the latest package from micropython.org, directly download GENERIC_S3-20220117-v1.18.bin (https://micropython.org/download/GENERIC_S3/), now this thonny can return the python character as shown below, I don’t know There is a problem there, see if you can help, generate a bin corresponding to the ESP32-S3-EYE integration, thank you
[image]
!

mocleiri, 2022-03-24:
Thanks for this report.  For esp32 and stm32 we have our pent board configuration that were copied from micropython upstream.

I suspect something was changed in the upstream configuration.

I don't have s3 boards but I can pick up the latest upstream changes and then trigger a build on that branch for you to test on.

pyharry, 2022-03-27:
Hello! Thank you very much for your reply and help! Looking forward to your builds for us to test, we'll try to provide feedback! At the same time, if it is convenient for you, I can provide an ESP32-S3-EYE v2.2 board to express and mail it to you to facilitate your joint debugging and build it for everyone to share, thank you!

mocleiri, 2022-04-15:
I found the change I made to the esp32s3's which was to disable USB.

I wonder if for your board or needed it to be there.

Are you setup to build from scratch?

You could try to revert this change and then see if the s3ima few then works for you:

https://github.com/mocleiri/tensorflow-micropython-examples/commit/2d0948b542c6c5bee92b2bb9e383fec59a1358f1

pyharry, 2022-04-17:
Hello! The resources we need to call include video, audio, SD card, USB, WIFI, Bluetooth, and ESP-NN acceleration in the official https://github.com/espressif/tflite-micro-esp-examples, please help, Generate and package a BIN for download testing, thank you!

pyharry, 2022-04-17:
Hello! The resources we need to call include video, audio, display, SD card, USB, WIFI, Bluetooth, and ESP-NN acceleration in the official https://github.com/espressif/tflite-micro-esp-examples, please For help, generate and package a BIN for downloading and testing, thank you!

pyharry, 2022-04-17:
Hello! The resources we need to call are video, audio, display, SD card, USB, WIFI, Bluetooth, and ESP-NN acceleration in the official https://github.com/espressif/tflite-micro-esp-examples, here The official C BIN can run, please help, generate and package a BIN called by micropython for downloading and testing, thank you!

mocleiri, 2022-12-19:
I took a more detailed look comparing each file from the MICROLITE_S3  project against the upstream GENERIC_S3 and found there was another change in the header file which is now applied. 

It enables repl over uart.

@pyharry can you check the applicable S3 build at this link have the new option set:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3727467156

I changed the build to assemble a single firmware.bin like upstream which is directly flashable.