mocleiri/tensorflow-micropython-examples #81: Didn't find op for builtin opcode 'SHAPE' version '1'

MATTYGILO, 2022-04-10:
When I load my model into the interpreter I get this error
```
Didn't find op for builtin opcode 'SHAPE' version '1'

Failed to get registration from op code SHAPE
 
Failed starting model allocation.
```

MATTYGILO, 2022-04-10:
```
Didn't find op for builtin opcode 'RANGE' version '1'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

Failed to get registration from op code RANGE
 
AllocateTensors() failed!
```

mocleiri, 2022-04-15:
I think this means your model is using an op not supported in tflite micro.

Here is the list of ops included: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc

There is no range op.

Look at you model in netron and double check each op is supported and if not then you need to adjust your model to only have compatible ops.

There may also be options to request the op to be added upstream or to port it to tflm yourself.

https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md