mocleiri/tensorflow-micropython-examples #90: install firmware

nicolashickl, 2022-06-23:
can you help me please, what is my mistake? I tried to install the esp32 firmware as a zip. now the following error message appears in thonny:
i have just connected a micro sd card adapter.

[image]

nicolashickl, 2022-06-24:
or is this the wrong way to install the firmware? (i just downloaded the zip file,, put it in thonny and installed it)

mocleiri, 2022-06-27:
You need to unzip the file as there are a couple of files within it.

The instructions I use from windows to flash are located on the main page: https://github.com/mocleiri/tensorflow-micropython-examples#flash-image

```
 esptool.py -p COM5 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 micropython.bin
```
on my system the esp32 shows up as COM5.

Someone posted the linux steps in one of the issues but I don´t recall which one at the moment but you can  search for it.

Be sure you have downloaded the image from the main branch as there are other branch builds which don´t work right.

nicolashickl, 2022-06-27:
thank you, now it works :)

mocleiri, 2022-08-29:
Closing as it is reported as working.

closed 2022-08-29 by mocleiri