mocleiri/tensorflow-micropython-examples #91: Both requests and urequests are missing

lcarnevale, 2022-08-09:
Hey there,

I wanna develop further applications on this firmware but both requests and urequests are missing.
How can I add them? What is the micropython version from which you started?

Lorenzo.

mocleiri, 2022-10-09:
@lcarnevale can you retest using the latest firmware which is on the latest micropython?

https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/2968453830

mocleiri, 2023-01-13:
I looked at one of the latest images and it doesn't have urequests either.

Its definitely manifest related.  Do you know of a board that has it?  I can try to understand what its doing and how to include the library into the firmware.

```
MicroPython d5f32c7-dirty on 2022-12-15; ESP32S2 Microlite with ESP32-S2FN4R2
Type "help()" for more information.
>>> help('modules')
__main__          gc                ubinascii         ure
_boot             inisetup          ucollections      uselect
_onewire          math              ucryptolib        usocket
_thread           microlite         uctypes           ussl
_uasyncio         micropython       uerrno            ustruct
_webrepl          neopixel          uhashlib          usys
apa106            network           uheapq            utime
btree             ntptime           uio               utimeq
builtins          onewire           ujson             uwebsocket
cmath             uarray            ulab              uzlib
dht               uasyncio/__init__ umachine          webrepl
ds18x20           uasyncio/core     uos               webrepl_setup
esp               uasyncio/event    upip              websocket_helper
esp32             uasyncio/funcs    upip_utarfile
flashbdev         uasyncio/lock     uplatform
framebuf          uasyncio/stream   urandom

```