mocleiri/tensorflow-micropython-examples #93: modules need update to work with new version of micropython

uraich, 2022-08-15:
The parameters in MP_REGISTER_MODULE have been modified in the latest micropython versions. The last parameter (enabling the module) has been removed.
This causes trouble when compiling. 
Also the micropython-lib must be initialized using make BOARD=... submodules 
I saw that master has not been updated since 6 months now.

mocleiri, 2022-08-17:
Thanks for specifying the steps required to fix the build.  I´m short of extra time in the last few months due to my day job.  

I´ll try to make time to get the build working again on the current tensorflow and micropython versions.

mocleiri, 2022-09-03:
I have a feature branch in progress where the builds are mostly working.  

I have a problem with the stm32 build due to changes in tflm for one of the cortex m optimized source files.  It needs access to one of the CMSIS header files and I'm trying to work out where that file lives in the micropython build and why it's not being included.

The latest micropython supports building with an external CMSIS which I may try to use to point the micropython build at what is downloaded by tflm.

I probably should make the fix for most of the boards and carry over the stm32 boys into a separate issue

closed 2022-10-06 by uraich