mocleiri/tensorflow-micropython-examples #98: Integrate microlite with OpenMV

tejalbarnwal, 2022-10-03:
Hey, is it possible to integrate the `microlite` with [OpenMV](https://openmv.io/)?

mocleiri, 2022-10-09:
The microlite module is sort of a fork of openmv.  The python API is different but I copied a lot of the glue code between micropython and tflm.

At the time this firmware started there was no OpenMV esp32 or rp2040 port.

If you have an stm32H7 you should use OpenMV.  For ESP32 you should consider microlite.

Both integrate tensorflow for micro controllers so if you have the model then you could deploy to either.

tejalbarnwal, 2022-10-11:
Hey, I am actually working with Arduino Nicla Vision, which kind of needs OpenMV to work.

tejalbarnwal, 2022-10-11:
Could you please guide me to where I could find the parts of microlite in the repository (specifically the regression examples) that follow the template of OpenMV?

mocleiri, 2022-10-11:
The micropython module on OpenMV is tf:
https://docs.openmv.io/library/omv.tf.html

This example shows how to flash OpenMV to your board:
https://docs.arduino.cc/tutorials/nicla-vision/blob-detection

Once it's flashed use the tf module to run your model.

closed 2022-10-11 by mocleiri