mocleiri/tensorflow-micropython-examples #99: ESP32 goes on a continuous printing loop after firmware for ESP32 is uploaded.

surajkumarpandey, 2022-10-05:
Hi all,
I was trying to upload the micropython TFlite firmware provided for ESP32  provided in the link "https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/2968453830"  (first one named as "microlite-esp32-firmware") but after uploading it, my thonny terminal goes into a non-stop loop printing following statements:

**"rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267148
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267148
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267148
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267148
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:267148
ets Jun  8 2016 00:22:57"**

I cannot save any file as it says 'Device is busy' nor can I make any other changes. Please aide over the matter.

uraich, 2022-10-05:
Which esp32 board are you using? Does it have PSRAM?

surajkumarpandey, 2022-10-05:
Thank you for responding. I have tried it with ESP EYE (8MB PSRAM) and ESP32-WROOM-32.

uraich, 2022-10-05:
Ok, Can you try to download and flash:
[firmware.bin](https://github.com/uraich/EdgeComputing/blob/main/firmware/firmware.bin)
and tell me if that works any better?

surajkumarpandey, 2022-10-05:
Thank you for the file but uploading this also yields a  continuous loop with following statements:

"ELF file SHA256: fc359ed41e6490b2

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13824
load:0x40080400,len:3340
entry 0x40080618
E (751) psram: PSRAM ID read error: 0xffffffff
E (751) cpu_start: Failed to init external RAM!

abort() was called at PC 0x400813c8 on core 0

Backtrace:0x40081ef1:0x3ffe3bb00x4008f679:0x3ffe3bd0 0x40095321:0x3ffe3bf0 0x400813c8:0x3ffe3c60 0x4007935e:0x3ffe3c90  |<-CORRUPTED

"

However, the original micropython firmware available at "https://micropython.org/download/esp32/" works fine and does not go into loop. I am able to program basic python with that.

surajkumarpandey, 2022-10-05:
No. I just use the update firmware option in Thonny to burn it as a new firmware.

surajkumarpandey, 2022-10-05:
Do we need to upload the bootloader and partioion-table together with the "micropython.bin" given. I am first uploading just the micropython file. Could that be an issue?

uraich, 2022-10-05:
No. In fact I am using an esp32-cam (much cheaper and less powerful than your board) Yours uses an esp32s3 chip while mine uses a standard esp32. As far as I can see, there is MICROLITE_S3_SPIRAM in tflite-micropython, but that does not include the camera driver. 
Therefore tflte-micropython should be compiled for the esp32s3 and the camera driver must be added. Since I don't have such a board, I can try to build an esp32s3 version but you would not be able ti use the camera.

surajkumarpandey, 2022-10-05:
I am for now experimenting with ESP32 WROOM which is a generic ESP32 chip without camera. I have kept ESP-EYE aside for now. But even on ESP32 WROOM the issue persists, which has the generic ESP32 chip.

uraich, 2022-10-05:
Does this board have PSRAM?

surajkumarpandey, 2022-10-05:
No it doesn't have PSRAM.

surajkumarpandey, 2022-10-05:
My ESP-EYE has esp32 chip only.

surajkumarpandey, 2022-10-05:
May I know how you uploaded the firmware?

surajkumarpandey, 2022-10-05:
Ok. It seems you did not use the partition-table and the bootloader as well. i will try to look out for more options. Still, thank you for your responses!

uraich, 2022-10-05:
I am using the makeimg.py from the esp32 micropython port to build firmware.bin from micropython.bin and that includes partition table and bootloader into the final firmware image. I do **not** flash micropython.bin but firmware.bin which is generated by makeimg.py.
On my esp32-cam this works without any problems.

surajkumarpandey, 2022-10-05:
It seems you are using ESP-IDF IDE.

surajkumarpandey, 2022-10-05:
It worked in windows so I am switching to windows for now. Thanks for the discussion!

surajkumarpandey, 2022-10-05:
Ok. I was using Ubuntu before. in windows I directly used the command given for uploading the firmware.

uraich, 2022-10-05:
I am using Ubuntu 22.04 myself. Did you try the shell scripts I gave you on Ubuntu? These should work, if you have a fairly new version of esp_tool installed.
Good luck with your further explorations.

surajkumarpandey, 2022-10-06:
Is there a documentation for the functions/objects used in the Hello_world example? I am trying to build a custom network that requires different configuration for running. For instance I wanted to know about the second parameter of "microlite.interpreter()" as to whose size is it and are callbacks optional for interpreter.

mocleiri, 2022-10-09:
@surajkumarpandey can you post the command you used to flash the firmware?

The firmware zip is not directly flash able.

The instructions on the readme show from the unzipped file how to flash the various parts:
https://github.com/mocleiri/tensorflow-micropython-examples#flash-from-windows

uraich, 2022-10-09:
Hi Michael,
I copied makeimg.py from the micropython sources to the boards directory of the board I am using (MICROLITE_SPIRAM_CAM) Then I created a simple shell script to call this 

#!/bin/sh

python3 makeimg.py \
build/sdkconfig \
build/bootloader/bootloader.bin \
build/partition_table/partition-table.bin \
build/micropython.bin \
build/firmware.bin \
build/micropython.uf2

This creates firmware.bin which you can easily flash

mocleiri, 2022-10-09:
@uraich thanks.

I'll incorporate that into the automatic builds to simplify flashing.

surajkumarpandey, 2022-10-09:
@mocleiri Thank you for asking. I flashed the firmware directly using thonny interpreter's 'update firmware functionality' in ubuntu (as I could not find any other available option). However, as I mentioned, I moved to windows later where it could be uploaded. But I wanted some help over the documentation of the functions used to be able to run a custom model.

mocleiri, 2022-10-09:
@surajkumarpandey thanks.  

I want to use this issue for changing the ESP32 builds to build a unified image suitable for direct flashing.

I'll move your other question about the API over to another issue where hopefully we can get you setup and improve the documentation for others in the future.

surajkumarpandey, 2022-10-09:
@mocleiri Sure! Thank you.

mocleiri, 2022-12-19:
Closing as the changes to create a single firmware.bin file using the upstream makeimg.py script has been merged into the main branch.  Merge commit is here:  2df65a06c2e27a38b3636eef497a4cdd04d7f964

closed 2022-12-19 by mocleiri