mocleiri/tensorflow-micropython-examples #103: Firmware for ESP32-C3

robostart-io, 2022-11-20:
Good day, dear Michael and creators of microlite!
Thanks for your wonderful project. This is a very handy item. It seems to me that his appearance is a very big event in the world of micropython.
Let me ask you, is it possible to create a firmware for the increasingly popular ESP32-C3 based on RISC-V?

mocleiri, 2022-11-27:
At  the moment I'm not building for all boards only the ones that people ask for and can verify.  If you have an ESP32-C3 to test with I can add a build for it.

Normally I copy the board configs from micropython and then customize as needed for the microlite module. Which board config are you using?

mocleiri, 2022-12-02:
I've started work on this but the build isn't quite working yet.  You may also want to take a look at https://github.com/espressif/tflite-micro-esp-examples

Which is Espressif's tflm repository, the only downside is you have to compile using C/C++ instead of using micropython.  

The C3 is only a single core so you may need the acceleration library integrated there to get acceptable speed.

Its a work in progress to get the esp_nn library merged into this repo.  I tried a few months ago but it locked up my normal esp32's when trying to run inference and I haven't been back around to try again.

robostart-io, 2022-12-02:
Thank you so much for getting started!
Unfortunately, working with ESP-IDF is problematic for many reasons for beginners.
So far, we are using your build for a regular ESP32 under MicroPython and are very happy with it.
We are ready to wait as long as necessary, and will make every effort to test and debug it.
Thank you very much again.

robostart-io, 2022-12-04:
I thought how I could be useful to your project? Decided to make a logo for it.
[Logo for Tensorflow Micropython Examples](https://github.com/robostart-io/ml-logo)

uraich, 2022-12-04:
Hmmm, Link does not work for me.

robostart-io, 2022-12-04:
Sorry, added access

mocleiri, 2022-12-10:
@robostart-io my latest change built the firmware successfully.  Can you download the applicable zip from here: https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3662193220

You have to unzip the file and then flash using the different parts.  This comment shows how the bits in the zip file can be combined into a flashable .bin file:

https://github.com/mocleiri/tensorflow-micropython-examples/issues/99#issuecomment-1272576632

There are a couple of more changes needed for the other ports before this work will merge down into the main branch.  But you should be able to confirm if the build is producing working firmware or not.

robostart-io, 2022-12-13:
Tried to flash from https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3666469707 and https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3662193220

by using esptool

python esptool.py erase_flash

python esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 micropython.bin

Connected and got a constantly repeating string

invalid header: 0xffffffff

Sometimes the following messages appear:

invalid header: 0xffffffff                                                                         
invalid headeюESP-ROM:esp32c3-api1-20210207                                                        
Build:Feb  7 2021                                                                                  
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)                                            
Saved PC:0x4004d1f8                                                                                
invalid header: 0xffffffff     
invalid header: 0xffffffff
...
invalid header: 0xffffffff
invalid header: 0xffffffff                                                                         
invalid header: 0шESP-ROM:esp32c3-api1-20210207                                                    
Build:Feb  7 2021                                                                                  
rst:0x10 (RTCWDT_RTC_RST),boot:0xc (SPI_FAST_FLASH_BOOT)                                           
invalid header: 0xffffffff
...

mocleiri, 2022-12-14:
Can you try flashing at a slower baud rate?  Sometimes that can help.

I found this esp-idf issue that had the same error message which might help: https://github.com/espressif/esp-idf/issues/9005

robostart-io, 2022-12-14:
Erased without problems

**python esptool.py erase_flash**
or
**python esptool.py -p COM13 --chip esp32c3 erase_flash**

```
PS H:\Electron\esptool\esptool-4.4> python esptool.py erase_flash
esptool.py v4.4
Found 2 serial ports
Serial port COM13
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:4d:cf:fc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.2s
Hard resetting via RTS pin...
PS H:\Electron\esptool\esptool-4.4> python esptool.py -p COM13 --chip esp32c3 erase_flash
esptool.py v4.4
Serial port COM13
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:4d:cf:fc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.1s
Hard resetting via RTS pin...
PS H:\Electron\esptool\esptool-4.4>
```

I tried changing speed on 115200 yesterday

Today I tried even lower, up to 10000 (at 9600 the esptool refuses to work with ESP32c3).

```
PS H:\Electron\esptool\esptool-4.4> python esptool.py -p COM13 --chip esp32c3 er
ase_flash
esptool.py v4.4
Serial port COM13
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:4d:cf:fc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.2s
Hard resetting via RTS pin...
PS H:\Electron\esptool\esptool-4.4> python esptool.py -p COM13 -b 10000 --before
 default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --
flash_size detect --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x8000 parti
tion_table/partition-table.bin 0x10000 micropython.bin
esptool.py v4.4
Serial port COM13
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:4d:cf:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x001fafff...
Compressed 15984 bytes to 10208...
Wrote 15984 bytes (10208 compressed) at 0x00001000 in 10.6 seconds (effective 12
.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 115...
Wrote 3072 bytes (115 compressed) at 0x00008000 in 0.2 seconds (effective 115.4
kbit/s)...
Hash of data verified.
Compressed 2008688 bytes to 1204669...
Wrote 2008688 bytes (1204669 compressed) at 0x00010000 in 1216.8 seconds (effect
ive 13.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
PS H:\Electron\esptool\esptool-4.4>
```

Changed the cable. Changed the board with ESP32c3. Changed the terminal program. The result is the same.

Yesterday I wrote: "After turning off the power and turning it back on, the COM port does not respond and does not output anything". This is not entirely true. After turning off the power and turning it on again, the port in some cases receives a constantly recurring `invalid header: 0xffffffff`.

How else could I test to help make the build?

robostart-io, 2022-12-14:
I use the following boards for testing

[ESP-C3-M1-Kit](https://docs.ai-thinker.com/_media/esp32/nodemcu-esp-c3-m1-kit_v1.2.0_specification.pdf)

[ESP-C3-32S(4M)-KIT](https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s-kit-v1.0_specification.pdf)

robostart-io, 2022-12-14:
I tried with old versions of esptool, but I get it again `invalid header: 0xffffffff`.

mocleiri, 2022-12-15:
What board config do you use from micropython?  For the microlite C3 images I am building directly from the micropython GENERIC_C3 and GENERIC_C3_USB configurations.

To troubleshoot I just made a new image for the esp32s2 as I do have this board https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-devkitm-1-v1.html

Mine got stuck at the power on step until I removed the usb configuration as this dev board has the serial chip.  Have you been flashing the generic_c3 image?

mocleiri, 2022-12-16:
Good catch, Can you try with the firmware starting at 0x0?  

I built the MICROLITE_C3 images directly using the upstream board config so any guidance on the placement should be the same.

It looks like its an esp32c3 particularity as some of the other c3 boards also say to flash from 0x0 instead of 0x1000 which is used for the xtensa based boards.

robostart-io, 2022-12-16:
If I write 0x0,

**python esptool.py -p COM6 -b 115200 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x0 micropython.bin**

then there is an overlap:

`esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: bootloader/bootloader.bin`

Maybe you need to do something with the partition-table?

robostart-io, 2022-12-16:
I tried to flash with the key **-z**

**python esptool.py -p COM6 -b 115200 --before default_reset --after hard_reset --chip esp32c3  write_flash -z --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 micropython.bin**

Got a recurring:

```
ESP-ROM:esp32c3-api1-20210207                                                                      
Build:Feb  7 2021                                                                                  
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)                                            
Saved PC:0x40047ed2                                                                                
SPIWP:0xee                                                                                         
mode:DIO, clock div:1                                                                              
load:0x3c1a0020,len:0x37880                                                                        
load:0x4d52203a,len:0x72642054                                                                     
Invalid image block, can't boot.                                                                   
ets_main.c 333                                                                                     
ESP-ROM:esp32c3-api1-20210207                                                                      
Build:Feb  7 2021                                                                                  
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)                                            
Saved PC:0x40047ed2                                                                                
SPIWP:0xee                                                                                         
mode:DIO, clock div:1                                                                              
load:0x3c1a0020,len:0x37880                                                                        
load:0x4d52203a,len:0x72642054                                                                     
Invalid image block, can't boot.                                                                   
ets_main.c 333
```

robostart-io, 2022-12-20:
I tried to write the bootloader to address 0x0 with the key -z and without it

**python esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 micropython.bin**

**python esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32c3 write_flash -z --flash_mode dio --flash_size detect --flash_freq 40m 0x0 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x10000 micropython.bin**

same as in previous post

mocleiri, 2022-12-21:
@robostart-io can you retry using the firmware available on the latest main branch:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3727394123

I integrated @uraich's script which builds a unified firmware binary that should be directly flashable using the micropython instructions.

Now there will be a single firmware.bin file in the zip file that can be flashed.

I tested flashing the standard esp32 microlite firmware and it worked so I think the c3 version should also work.

```
esptool.py --chip esp32c3 --port COM6 --baud 460800 write_flash -z 0x0 firmware.bin
```

robostart-io, 2022-12-23:
So far, unfortunately, it hasn't worked.
Outputs repeated:

```
ELF file SHA256: 7c88102a732dcb9b                                                                                                                                           
                                                                                                                                                                            
Rebooting...                                                                                                                                                                
ESP-ROM:esp32c3-api1-20210207                                                                                                                                               
Build:Feb  7 2021                                                                                                                                                           
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)                                                                                                                     
Saved PC:0x403806fc                                                                                                                                                         
SPIWP:0xee                                                                                                                                                                  
mode:DIO, clock div:1                                                                                                                                                       
load:0x3fcd6100,len:0xe3c                                                                                                                                                   
load:0x403ce000,len:0x6f4                                                                                                                                                   
load:0x403d0000,len:0x28ec                                                                                                                                                  
entry 0x403ce000                                                                                                                                                            
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.                                                                                       
                                                                                                                                                                            
Core  0 register dump:                                                                                                                                                      
MEPC    : 0x4039071c  RA      : 0x40390734  SP      : 0x3fcc3e00  GP      : 0x3fc93600                                                                                      
TP      : 0x3fc8da0c  T0      : 0x403915a4  T1      : 0x0000000f  T2      : 0x00000000                                                                                      
S0/FP   : 0x7f8d9224  S1      : 0x7f8d9220  A0      : 0x3fc9f81c  A1      : 0x3fc39a08                                                                                      
A2      : 0x00000000  A3      : 0x3fcc3e44  A4      : 0x00000000  A5      : 0x3fc39a08                                                                                      
A6      : 0x3fcc3e70  A7      : 0x00000000  S2      : 0x4039135c  S3      : 0x3fcc3e44                                                                                      
S4      : 0x00000004  S5      : 0x000437a0  S6      : 0x00000000  S7      : 0x00000000                                                                                      
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000                                                                                      
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000                                                                                      
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x7f8d9224                                                                                      
MHARTID : 0x00000000                                                                                                                                                        
                                                                                                                                                                            
Stack memory:
3fcc3e00: 0x3fcc4cb0 0x00000000 0x00000000 0x00000000 0x00000004 0x3fcc3e44 0x3fc9d090 0x403915c2                                                                           
3fcc3e20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000004 0x3fcc3e74 0x3fc9d770 0x4207af5a                                                                           
3fcc3e40: 0x3fcc3e70 0x00000000 0x00000000 0x3fc39a08 0x00000000 0x00000004 0x00000001 0x00000005                                                                           
3fcc3e60: 0x00000000 0x00000000 0x00000000 0x4207afb2 0x3fcc5108 0x00000000 0x00000000 0x00000000                                                                           
[... cut ...]
3fcc4180: 0x3c1af23c 0x420747dc 0x3c1af23c 0x420747d2 0x3c1af23c 0x420747c8 0x3c1af23c 0x420747be                                                                           
3fcc41a0: 0x3c1af23c 0x420747b4 0x3c1af23c 0x420747aa 0x3c1af23c 0x420747a0 0x3c1af23c 0x42074afe                                                                           
3fcc41c0: 0x3c1af23c 0x42074796 0x3c1af23c 0x4207478c 0x3c1af23c 0x420752b0 0x3c1af23c 0x42074782                                                                           
3fcc41e0: 0x3c1af23c 0x42074778 0x3c1af23c 0x4207476e 0x3c1af23c 0x42074764 0x3c1af278 0x42074918
```

mocleiri, 2023-01-13:
I bought an esp32c3 board to further investigate this issue.  https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html

I was able to successfully flash micropython 1.19.1 but get the same load access fault error when booting on microlite.

I need to look at the espressif tflite-micro repo to see if they are doing anything special for esp32c3.  On the tensorflow side the generator isn't doing anything xtensa specific.

There is a way to decode the fault or I might need to solder the JTAG pins to debug in visual studio to try and understand where the error is coming from.
```
90648  RA      : 0x40390660  SP      : 0x3fcc3e00  GP      : 0x3Guru Meditation                                                                         Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40390648  RA      : 0x40390660  SP      : 0x3fcc3e00  GP      : 0x3f                                                                        c93600
TP      : 0x3fc8db2c  T0      : 0x403914d0  T1      : 0x0000000f  T2      : 0x00                                                                        000000
S0/FP   : 0x7f8d91f4  S1      : 0x7f8d91f0  A0      : 0x3fc9f7ec  A1      : 0x3f                                                                        c39a08
A2      : 0x00000000  A3      : 0x3fcc3e44  A4      : 0x00000000  A5      : 0x3f                                                                        c39a08
A6      : 0x00000000  A7      : 0x3fcc0704  S2      : 0x40391288  S3      : 0x3f                                                                        cc3e44
S4      : 0x00000004  S5      : 0x000437d0  S6      : 0x00000000  S7      : 0x00                                                                        000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00                                                                        000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00                                                                        000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x7f                                                                        8d91f4
MHARTID : 0x00000000

Stack memory:
3fcc3e00: 0x3fcc4ca8 0x00000000 0x00000000 0x00000000 0x00000004 0x3fcc3e44 0x3f                                                                        c9d060 0x403914ee
3fcc3e20: 0x00000009 0x00000081 0x00000002 0x00000000 0x00000004 0x3fcc3e74 0x3f                                                                        c9d740 0x42079f84
3fcc3e40: 0x00000000 0x00000000 0x00000000 0x3fc39a08 0x00000000 0x00000004 0x00                                                                        000001 0x00000005
3fcc3e60: 0x00000000 0x00000000 0x00000000 0x42079fdc 0x00000000 0x00000000 0x00                                                                        000000 0x00000000
3fcc3e80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000000 0x420aaf48
3fcc3ea0: 0x00000000 0x3fcc3ea0 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000000 0x00000000
3fcc3ec0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000000 0x00000000
3fcc3ee0: 0x00000000 0x00000000 0x00000000 0x4038e2fe 0x00000000 0x00000000 0x00                                                                        000000 0x00000000
3fcc3f00: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3c1af628 0x3c1a01d4 0x3c                                                                        1af188 0x42000422
3fcc3f20: 0x3c1af1c4 0x420003e4 0x3c1af1c4 0x42000150 0x3c1af1c4 0x4200056e 0x3c                                                                        1af188 0x420004de
3fcc3f40: 0x3c1af1c4 0x4200026e 0x3c1af1c4 0x42000362 0x3c1af200 0x420002e6 0x3c                                                                        1af1c4 0x420001fe
3fcc3f60: 0x3c1af1c4 0x420001da 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000001 0x00000000
3fcc3f80: 0xfffffff7 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000002 0x00000001
3fcc3fa0: 0x00000000 0x00000003 0x00000000 0x00000002 0x00000001 0x00000000 0x00                                                                        000000 0x00000002
3fcc3fc0: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000000 0x00000000
3fcc3fe0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00                                                                        000001 0x00000000
3fcc4000: 0x00000000 0x00000004 0x00000003 0x00000002 0x00000001 0x00000000 0x02                                                                        03035a 0x00000912
3fcc4020: 0x3c1af188 0x4205d258 0x3c1af1c4 0x420608f8 0x3c1af188 0x4205d764 0x3c                                                                        1af188 0x4205d5c0
[... cut ...]
3fcc4180: 0x3c1af188 0x420747dc 0x3c1af188 0x420747d2 0x3c1af188 0x420747c8 0x3c                                                                        1af188 0x420747be
3fcc41a0: 0x3c1af188 0x420747b4 0x3c1af188 0x420747aa 0x3c1af188 0x420747a0 0x3c                                                                        1af188 0x42074afe
3fcc41c0: 0x3c1af188 0x42074796 0x3c1af188 0x4207478c 0x3c1af188 0x420752b0 0x3c                                                                        1af188 0x42074782

```

robostart-io, 2023-01-13:
I am very glad that you now also have an ESP-C3 board. Thank you so much for your serious efforts.