mocleiri/tensorflow-micropython-examples #105: Problem about firmware for Raspberry Pi Pico

Feihong-web, 2022-12-15:
I have drag the firmware.uf2 file to my Raspberry Pi Pico broad. But my computer and THonny IDE can not recognize the broad as an USB device correctly and I can not program on the Pico broad.

mocleiri, 2022-12-17:
Can you try to put the board into flash mode before moving the .uf2 file over?

https://micropython.org/download/rp2-pico/

It says to hold down the boot select button as you plug in the usb cable and that will present the disk where it will flash the .uf2 file when copied.

If you just copy it over you are doing a file copy not flashing a new firmware.

mocleiri, 2023-01-13:
Can you try flashing the image for **microlite-rp2-adafruit-feather-rp2040-firmware**

https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/3727394117

That's the board I have and others reported that the normal image doesn't work but they were able to flash this one.

dattran-itrvn, 2023-06-28:
This error is related to 'MICROPY_HW_FLASH_STORAGE_BASE' in the 'micropython/ports/rp2/rp2_flash.c' file. You need to reduce the size of 'MICROPY_HW_FLASH_STORAGE_BYTES' in the '\micropython\ports\rp2\boards\PICO\mpconfigboard.h' file.
`#define MICROPY_HW_FLASH_STORAGE_BYTES          (1024 * 1024)`