mocleiri/tensorflow-micropython-examples #109: Updating to the new version of micropython

MATTYGILO, 2023-02-28:
Could you update the micropython version, because there are a lot of bug fixes that I require that are only available in the newest version of micropython

mocleiri, 2023-03-11:
The current build is annoying because it doesn't notice upstream changes in any of the sub-modules. 

In the tensorflow micro side most projects are embedding the latest snapshot of the tflm code using a code sync job that pulls changes in from the upstream into the examples project.

In ulab they do the opposite and don't have submodules and the build script clones the micropython repository and then the code stands alone.

I'm learning towards the ulab approach but using a shallow clone to speedup the build.
https://github.com/v923z/micropython-ulab/blob/master/build.sh#L46

For local development it would still allow you to have the full clone of the other repos.

mocleiri, 2023-05-05:
Should update micropython to version 1.20