mocleiri/tensorflow-micropython-examples #112: tensor object has no attribute getInputTensor

lcarnevale, 2023-03-28:
Hey there,

I am running the hello_world project on the latest firmware available for the ESP32.
I use ampy to run the main and the application fails with this issue:

`ampy.pyboard.PyboardError: ('exception', b'interpreter_make_new: model size = 2488, tensor area = 2064\r\ntime step,y\r\n', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\n  File "hello.py", line 60, in <module>\r\n  File "hello.py", line 17, in input_callback\r\nAttributeError: \'tensor\' object has no attribute \'getInputTensor\'\r\n')`

horvathgergo, 2023-04-15:
Same issue here. I'm using Thonny IDE and ESP32 trying to run micro_speech:

interpreter_make_new: model size = 18712, tensor area = 8208
Starting
caught exception AttributeError 'tensor' object has no attribute 'getInputTensor'
Done
Traceback (most recent call last):
  File "<stdin>", line 344, in <module>
  File "<stdin>", line 317, in runModel
  File "<stdin>", line 45, in input_callback
AttributeError: 'tensor' object has no attribute 'getInputTensor'

AbdouTobblaze, 2023-04-18:
Hi, I got the same issue : AttributeError: 'interpreter' object has no attribute 'getInputTensor' when running the hello-world example on the ESP32S3.

interpreter_make_new: model size = 2488, tensor area = 2064
time step,y
Traceback (most recent call last):
  File "<stdin>", line 60, in <module>
  File "<stdin>", line 17, in input_callback
AttributeError: 'tensor' object has no attribute 'getInputTensor'

Can you please fix it

mocleiri, 2023-05-05:
I have an experimental branch where I reverted changes to go back to the openmv baseline for how inference works to remove a single dynamic memory allocation which has been causing issues with the stm32 port and I thought also the esp32c3 port but It didn't fix it.

It changes the lifetime of the interpreter from being heap based to being stack based so the Tensor object no longer exists outside of the callback functions.

The examples work with images from the main branch here:
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/4791546929 (esp32)
https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/4791546936 (esp32s3)