mocleiri/tensorflow-micropython-examples #121: Requesting support for ESP32-S3 with camera module

tarchive, 2023-06-09:
As pointed out in another issue, there are two variants of the ESP-EYE. https://github.com/mocleiri/tensorflow-micropython-examples/issues/96#issuecomment-1269156018

 Currently this repo only covers the ESP32 variant with camera support and so I am requesting support also be added for the ESP32-S3 variant. 
 
[image]

vikramdattu, 2023-07-20:
@tarchive did you try with `MICROLITE_SPIRAM_CAM` board with `idf.py set-target esp32s3`?

uraich, 2023-08-29:
Could you please try this:
https://github.com/uraich/EdgeComputing/blob/main/firmware/firmware-esp32s3-cam.bin
[image]
Just managed to compile it but did not do any testing apart from importing the modules

Henry1iu, 2023-11-21:
> Could you please try this: https://github.com/uraich/EdgeComputing/blob/main/firmware/firmware-esp32s3-cam.bin [image] Just managed to compile it but did not do any testing apart from importing the modules

I have tested it already and found no issues.

whittenator, 2023-12-13:
> Could you please try this: https://github.com/uraich/EdgeComputing/blob/main/firmware/firmware-esp32s3-cam.bin [...]

How did you build this binary? My FireBeetle 2 ESP32-S3 crashes every time I call camera.init().

uraich, 2023-12-14:
Hi Travis,
The camera driver expects this connection map:
[image]
(see micropython-modules/micropython-camera-driver/src/modcamera.h)
If your board has different connections (which I suspect), then you must specify these when initializing the camera.
I have a Freenove ESP32S3 board with camera for which the initialization looks like this:
[image]
Hope this helps

whittenator, 2023-12-14:
> Hi Travis,
>
> The camera driver expects this connection map:
>
> [image]
>
> (see micropython-modules/micropython-camera-driver/src/modcamera.h)
>
> If your board has different connections (which I suspect), [...]

Thanks for the speedy reply!

I am using the DFRobot FireBeetle 2 ESP32-S3 that comes with the OV2640. I am doing the camera.init(0) that is recommended for my camera. Any ideas on how I can see why it restarts the board whenever I do this camera.init(0) call?

uraich, 2023-12-14:
As I said before, please check that the hardware connections of your camera correspond to connections of the first screen dump. Do they?