mocleiri/tensorflow-micropython-examples #122: Improvements to the Repo

MATTYGILO, 2023-06-23:
More than happy to help with these, but I think it is important we focus on these improvements

1. Update to the repo to the newest micropython
2. Include ESP_NN

MATTYGILO, 2023-06-23:
I have begun to update the code to the newest micropython version and updating tflite-micro but I am currently getting these errors:
```
microlite/openmv-libtf.cpp:6:10: fatal error: tensorflow/lite/micro/all_ops_resolver.h: No such file or directory
 #include "tensorflow/lite/micro/all_ops_resolver.h"
```

MATTYGILO, 2023-06-23:
So the reason behind this issue is since I now implement tflite micro, it no longer has all_ops_resolver.h in "tensorflow/lite/micro/all_ops_resolver.h"

[image]

MATTYGILO, 2023-06-23:
So I fixed this error by converting it to 
`#include "tensorflow/lite/micro/micro_mutable_op_resolver.h"`

MATTYGILO, 2023-06-24:
Question: Why do we convert .cc files to .cpp

mocleiri, 2023-06-24:
#120 is going to address items 2 and 3 from your list.

> Question: Why do we convert .cc files to .cpp

Micropython assumed .cpp files for C++ modules.  Originally when the build was based on Gnu Make the QSTR generation was collecting all of the .cpp files.  So the renaming was done to get the QSTR generation to work properly.

TFLM uses .cc for all of the source files.

I think for the esp32/rp build the use of CMake changes how the sources are collected to not care about the extension.

The preparation script [micropython-modules/microlite/prepare-tflm-esp.sh](https://github.com/mocleiri/tensorflow-micropython-examples/blob/main/micropython-modules/microlite/prepare-tflm-esp.sh) generates the tflm base files and uses the --rename_cc_to_cpp option so that the generated files have .cpp extensions.

closed 2023-06-30 by MATTYGILO