mocleiri/tensorflow-micropython-examples #125: Micropython Upgrade to the latest

kyeonju23, 2023-08-23:
I want to build a firmware using the latest version of micropython 1.20.0, esp-idf v5.0.2 and the latest version of ulab. 
I tried to change it separately, but the firmware creation keeps failing. Can you make a newer version of the esp32 firmware?

uraich, 2023-08-25:
I have just started to attempt exactly this. Do you just need the binary?

kyeonju23, 2023-08-26:
Hi. First of all, I need a binary, and I want to know how to make a firmware. This is because the latest version, including ulab, was successful, but tflite-micro and examples continued to fail.

uraich, 2023-08-29:
Hi @kyeonju23 
can you please tell me on which board you want to run the interpreter. Processor type: esp32 esp32s2,s3,...
Myself I only have the esp32 and the esp32s3.
Does your board have a camera

kyeonju23, 2023-09-01:
It uses an esp32 board and has no camera.

uraich, 2023-09-01:
Ok, I uploaded a firmware version for the esp32, no spiram, no camera to github:
https://github.com/uraich/EdgeComputing/tree/main/firmware/firmware-esp32.bin
Can you please download and flash it and tell me if it works for you?

uraich, 2023-09-04:
Any luck?

uraich, 2023-09-05:
Hmmm, I don't understand!
I just downloaded the firmware and flashed it onto an ESP32, no SPIRAM, no camera. I connect to the ESP32 via the serial line and minicom.
[image]
As you can see, I can import microlite without problems.

kyeonju23, 2023-09-05:
sorry. I've been doing the import wrong so far. I used the method you suggested and the import was successful. thank you.

>>> import microlite
>>>

kyeonju23, 2023-10-05:
hello.
Thanks to the binary files you sent me, the project is progressing well.
thank you.

As the project progresses, there are more things I want to add. If it's
okay, can I receive the binary file creation process as a yml file or code?